LumaTeam/Luma3DS #1800: Missing options in luma3ds menu

Moepkid, 2022-09-25:
**System model:**

New 2ds XL

**SysNAND version (+emu/redNAND version if applicable):**

Latest (49E)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

V11.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

I'm missing the following three options from my menu:

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

Should these not be present? I Had to reinstall my luma because of a factory reset. Im worried I miss essential stuff.

Shadow322, 2022-09-25:
> **System model:**
> 
> New 2ds XL
> 
> **SysNAND version (+emu/redNAND version if applicable):**
> 
> Latest (49E)
> 
> **Entrypoint (How/what you're using to boot Luma3DS):**
> 
> Boot9Strap
> 
> **Luma3DS version:**
> 
> V11.0
> 
> **Explanation of the issue:**
> 
> I'm missing the following three options from my menu:
> 
> Set developer UNITINFO: ( )
> 
> Disable Arm11 exception handlers: ( )
> 
> Enable Rosalina on SAFE_FIRM: ( )
> 
> Should these not be present? I Had to reinstall my luma because of a factory reset. Im worried I miss essential stuff.
> 
> I also cant create a emunand backup through god9mode. So im worried I messed up the install

reading does a body good

[image]

Moepkid, 2022-09-27:
Im sorry. Im completely new to this :(

closed 2022-09-27 by Moepkid