LumaTeam/Luma3DS #1807: [Suggestion] allow locale emulation on applets

cooolgamer, 2022-10-28:
Hello, I noticed that by editing the line that prevent some apps from using game patching allow locale emulation working fine on applets.
Would it be possible to allow applets to use locale emulation? Thanks

PabloMK7, 2022-11-14:
Can you give examples of applets that would benefit from locale emulation?

cooolgamer, 2022-11-19:
I first tried with the keyboard and it work just fine, It's useful because russian one have a extra keyboard tab
also instruction manual, friend list, game notes etc. do work too

cooolgamer, 2022-11-19:
the problem is I don't really know how to code xD
I only know how to edit some things but I don't know how to make it allow applets to ONLY use locale emulation patch as I enabled all patches for applets for testing

cooolgamer, 2023-02-10:
done in v12.0

closed 2023-02-10 by cooolgamer