LumaTeam/Luma3DS #1808: 2DS Exception error

MrPotatoisgod, 2022-10-30:
**System model: Old 2DS**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable): 11.16.0-47E**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS): Boot9Strap**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version: 11.0**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options: Default**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
Boot after installation gives an error "An exception occurred".
Clearing ID0 and Nintendo 3DS folders do nothing.

**Steps to reproduce:**

1. Install Luma3DS on a 2DS

2. Boot the 2DS up

**Dump file: (https://github.com/LumaTeam/Luma3DS/files/9896399/crash_dump_00000000.log)-- Rename extension to ".dmp" as I couldn't upload it otherwise**

Gruetzig, 2022-10-30:
I helped them a bit, let me add that it boots fine in Stock Mode(used safety test)

MrPotatoisgod, 2022-10-30:
Oh, yeah. Forgot that :¬/