LumaTeam/Luma3DS #1809: Disable Errdisp and replace it with this

CZ2746isback, 2022-11-04:
[image]

Masamune3210, 2022-11-05:
IMO sounds like a really good way to get people to start putting in a lot of panic issue reports because they see scary errors and like to jump to conclusions

CZ2746isback, 2022-11-05:
> IMO sounds like a really good way to get people to start putting in a lot of panic issue reports because they see scary errors and like to jump to conclusions

I mean, this one is more detailed

CZ2746isback, 2022-11-07:
> IMO sounds like a really good way to get people to start putting in a lot of panic issue reports because they see scary errors and like to jump to conclusions

Oh wait... Developer Crash Handler can be confused with the SM64 one.

PabloMK7, 2022-11-14:
The reason Luma has its own error handler is for freeing all the memory that is neeeded by the official error handler. That extra memory is then taken by Rosalina. If you add the official error handler back, then there may be memory issues with Rosalina.

I also agree that the exception handler is "scary", may be worth reworking it to make it more user friendly.

CZ2746isback, 2022-11-16:
> The reason Luma has its own error handler is for freeing all the memory that is neeeded by the official error handler. That extra memory is then taken by Rosalina. If you add the official error handler back (or the developer version), then there may be memory issues with Rosalina.
> 
> I also agree that the exception handler is "scary", may be worth reworking it to make it more user friendly.

Oh, well that's interesting... I personally think that the exception handler is also scary since it pops out of nowhere (after a black screen, on a silent screen)

profi200, 2022-11-16:
I have been disabling the exception handler on my gaming 3DS personally. Not because it doesn't work or anything but because it does a rather abrupt reboot giving HOS no time to terminate sysmodules. I already had fs corruption occur because of this. For homebrew development it should absolutely stay on.

As for errdisp:
It makes no sense to use Nintendos version imho because it provides less information and is more bloated than Lumas replacement.

CZ2746isback, 2022-11-16:
> I have been disabling the exception handler on my gaming 3DS personally. Not because it doesn't work or anything but because it does a rather abrupt reboot giving HOS no time to terminate [...]

1. I had to fix something because of that.
2. What do you mean by "bloated"?

Masamune3210, 2022-11-16:
It uses way more resources than it really should need to. It's a error dialog, it doesn't have to be gigantically flourish filled and eat memory like popcorn

profi200, 2022-11-16:
>     1. I had to fix something because of that.
> 
>     2. What do you mean by "bloated"?
What has been described above. ErrDisp is its own sysmodule and it wastes lots of memory which can be freed for other purposes.

closed 2022-12-04 by suppai