LumaTeam/Luma3DS #1811: clock speed automatically going to 800mhz on games that dont work well with it

kamikazilucas, 2022-11-07:
im trying to play animal crossing but everytime i play the game it goes to 800mhz which makes the game buggy, can there be an option to have it default to 200mhz instead its soo annoying having to to into the menu every time i start the game

PabloMK7, 2022-11-14:
If you see the game going to 800mhz it's because the game itself is asking the system to go to 800mhz. I doubt that is causing the game to be buggy as that's what the game is designed to.

kamikazilucas, 2022-11-14:
why is it supposed to go to 800mhz when its an og 3ds game, did it get new 3ds support?

PabloMK7, 2022-11-14:
Are you sure you are not confusing the UI? What does the new 3ds configuration menu say while you are on the home menu without any games opened? 200mhz or 800mhz?

kamikazilucas, 2022-11-14:
home menu is 800mhz, but when i open the game its also 800mhz with cache turned on

PabloMK7, 2022-11-14:
Check that you didn't force 800mhz mode in the Luma configuration menu (hold select on boot)

PabloMK7, 2022-11-14:
Which animal crossing game is it?

kamikazilucas, 2022-11-14:
new leaf

PabloMK7, 2022-11-14:
Are you using a 3GX plugin/CTRPF?

kamikazilucas, 2022-11-14:
no i dont think so just using luma

profi200, 2022-11-14:
Can't reproduce. It says "Set clock to 268 MHz". L2 cache is also off. The game is fully updated.

kamikazilucas, 2022-11-14:
if it says set to 268 it means its running at 800mhz

profi200, 2022-11-14:
The game is running as expected. Zero glitches. They probably switched to New 3DS clock with the welcome Amiibo update.

kamikazilucas, 2022-11-14:
yes except it uses more power and has problems saving when wireless is an issue

PabloMK7, 2022-11-14:
I also checked, the game runs at the new 3ds speed because they updated the game to support that, so it's not a bug with Luma.

kamikazilucas, 2022-11-14:
ok thanks, well ig ill just cope with it then, i wish the wireless wasent so annoying though idk why its doing that

closed 2022-12-04 by suppai