LumaTeam/Luma3DS #1812: An error occurred (ErrDisp)

Err-999-Not-Here, 2022-11-11:
I have just recently installed CFW on my Nintendo 3ds LL for the purpose of unlocking the region. In successfully doing so, I found that every time I insert a game card to my 3ds, it gives me an error code. I will copy and paste it below. The strange thing is, It only works on physical games. I have many digital games and they all work just fine. I followed the 3ds.guide tutorial for installing CFW. Here is the error code:
Error Type: Generic
Process ID: 19
Process Name: am
Process Title ID: 0x0004013000001502
Address: 0x001093aa
Error Code: 0xc8804478
Press any button to reboot.

Masamune3210, 2022-11-11:
Go back and fill out the issue template

Err-999-Not-Here, 2022-11-11:
> 

What do you mean?

Masamune3210, 2022-11-11:
Sorry, used to Atmosphere where there is a mandatory issue template. I apologize

PabloMK7, 2022-11-14:
This is probably unrelated to an issue in Luma3DS (this page is for reporting issues with Luma3DS and not for support). Can you please close this issue and join the Homebrew Discord server and ask for help there? https://discord.gg/C29hYvh

closed 2022-12-04 by suppai