LumaTeam/Luma3DS #1813: Issue with Game Card Power toggle

BlakeEatsRamen, 2022-11-14:
I've noticed that there's an issue with the game card slot power toggle that prevents game cards from being read at all. I uninstalled Luma to test my theory and I believe this is correct. Please look into this and address it as needed

Device: Nintendo 2ds running latest firmware
Luma: V.11
Applicable games (that I've tested): Monster Hunter Stories, Tomodachi Life, MarioKart 7, Super Smash Bros. For 3ds, Shin Megami Tensei IV

GloriaVermillion, 2022-11-14:
It doesn't work for me
Can you kindly tell me what system you're using?

profi200, 2022-11-14:
Can't reproduce. It works as advertised. Turn gamecard power back on in the Rosalina menu before re-inserting the game.

GloriaVermillion, 2022-11-14:
Okay, I will try that. Apologies for wasting your time