LumaTeam/Luma3DS #1814: can't disable arm11 exception handlers (Option Missing)

KingDoogan, 2022-11-14:
[image]

PabloMK7, 2022-11-14:
Do you have any specific reason to disable arm11 exception handlers? Unless you are a developer you shouldn't be disabling them (that's the main reason they were removed from the built-in config editor). If you know what you are doing and still want to disable them, edit the config.ini file in the luma folder in your SD.

KingDoogan, 2022-11-15:
> Works as intended.
> 
> Do you have any specific reason to disable arm11 exception handlers? Unless you are a developer you shouldn't be disabling them (that's the main reason they were removed from the built-in config editor). If you know what you are doing and still want to disable them, edit the config.ini file in the luma folder in your SD.

Doing some testing with my unity 3ds project. (Only solution i've found, is disabling this option)

Thanks!

closed 2022-11-15 by KingDoogan