LumaTeam/Luma3DS #1815: Unable to load ultra sun/ ultra moon

Jmc1922, 2022-11-17:
I am using the lastest version of luma3ds and my Nintendo switch is up to date . I can play omega ruby fine but when I go to play ultra sun/moon I always get errors 

[image]

Is there any way of helping

Masamune3210, 2022-11-17:
Cart or digital? If digital, redownload from eshop

Masamune3210, 2022-11-17:
If the game works without your attempt at modding, I would assume something went wrong with making the mod and it's not valid

PabloMK7, 2022-11-20:
This is probably unrelated to an issue in Luma3DS (this page is for reporting issues with Luma3DS and not for support). Can you please close this issue and join the Homebrew Discord server and ask for help there? https://discord.gg/C29hYvh

closed 2022-12-04 by suppai