LumaTeam/Luma3DS #1824: 3ds bricked?

cheryyB, 2022-12-28:
I was needing to find a save file and to do so tutorials made me downgrade to version 2.1. when I did what the tutorials told me I got a pop up of the exception arm11 dump. When I tried fixing this it said to download luma again then disable arm11 and I don't get the dump pop anymore but when I boot it will instead show a black screen with the solid blue light and no matter how long I wait it just stays black. I can still open luma and gm9 but not the home menu any help is very appreciated

urherenow, 2022-12-28:
Github is not a support forum, it’s for collaboration and reporting bugs. There are tutorials on gbatemp for this. You could try to google how to get into recovery and attempt to install firmware that way…

cheryyB, 2022-12-28:
ok thank you

closed 2023-01-28 by suppai