LumaTeam/Luma3DS #1825: Constant crashes with certain games

SberryfieldS, 2022-12-30:
Model: New 2DS XL
Region: LatAm
Problem: Some games have been rebooting the console on their own, namely Tekken Prime Edition (only happened once), Mario Kart 7 (only when racing online) and Smash Bros. (happpened three times halfway through gameplay). I should point out that I've installed Mario Kart 7's 1.2 update a few days ago, I heard they patched an exploit in the game that allowed you to install homebrew, but I'm not sure that might have something to do with the issue. Jrpgs such as any of the Fire Emblems, Xenoblade nor Pokémon seem to replicate this issue.
Nand: Sys 11.16.0-49U
Processor: Arm11 (core 0)
Current process: 

**Entrypoint**
Boot9strap 

**Luma3DS version:**
11 

**Luma3DS configuration/options:** 

Autoboot EmuNAND: ( )
 

Use EmuNAND FIRM if booting with R: ( )
 

Enable loading external FIRMs and modules: ( )
 

Enable game patching: (X) 

Show NAND or user string in System Settings: ( ) 

Show GBA boot screen in patched AGB_FIRM: ( ) 

Set developer UNITINFO: ( ) 

Disable Arm11 exception handlers: ( ) 

Enable Rosalina on SAFE_FIRM: ( )
 

**Steps to reproduce:**
For Tekken and Smash Bros.: happens while playing the game, not on menus, only when  fighting. In Tekken's case, the game crashed and immediatly went on to the luma boot menu. It hasn't happened again ever since with that game. Smash Bros., on the other hand, has crashed every single time. I've only tested mario kart once.

**Dump file:**
There's no dump that I'm aware of. I'm posting a picture of the error: 

https://ibb.co/dtWzZSg

Masamune3210, 2022-12-31:
is this on a o3DS? Could be related to a issue with the memory mode that some bigger games use which soft reboots the system to shuffle memory around

TuxSH, 2023-01-10:
The error corresponds to "SD card not found/missing", which is triggered when you remove the SD card while playing contents installed on your SD card. Check your sd card and sd port, this is not a Luma issue.

closed 2023-01-23 by TuxSH