LumaTeam/Luma3DS #1826: BootNTRSelector for some reason doesn't work/Luma3DS supporting mapping other processes' individual memory pages.

ZGStuff, 2023-01-02:
I want to use NTR Capture software for Luma (I'm on 10.2.1 because in my opinion it's the most stable), but when I go to [BootNTRSelector by Nanquitas](https://github.com/Nanquitas/BootNTR), it says on the latest release that official Luma3DS is broken with it. I am NOT a Homebrew expert, and REALLY don't want to have to uninstall and reinstall Luma to use capture.

So. Is this INTENTIONAL, or am I just an idiot and this was fixed in 11.0?

ZGStuff, 2023-01-02:
it wasn't fixed in 11.0 *sob*

urherenow, 2023-01-02:
Gbatemp is where you want to ask for help, preferably after searching the forums first. Even discord. But github is not a support forum.

One doesn’t “uninstall and reinstall” Luma. You just drop in the new files from the release, overwriting the old files… so you don’t want that last question answered…

ZGStuff, 2023-01-02:
ah
to Luma3DS 3GX Loader i change to!

closed 2023-01-02 by ZGStuff