LumaTeam/Luma3DS #1830: O3ds won't start

ta-gueule, 2023-01-11:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

Can't access to system options

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v11.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (off)

Splash duration: ()

PIN lock: (off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
After put my o3ds in sleep mode while playing tomodachi life (eur version), I opened my o3ds 15 minutes left, and I had a black screen with just the music of the game, I rebooted my device and 
I had a error screen (an exception occurred) for prefetch abort (svcBreak).
I can't access to home menu or anything else, I just can access to luma start menu and gamemode9.
I tried ctrTransfer and it didn't work, I also tried to replace boot.firm and boot.3ds files on the root
of the sd cart, didn't work too.

Masamune3210, 2023-01-11:
Something smells like SD corruption or issues with the SD reader

ta-gueule, 2023-01-14:
> 

Yes i thought

ta-gueule, 2023-01-14:
> Something smells like SD corruption or issues with the SD reader

Yes i thought that was a corruption too but i replaced the 2 boot files of the root of the sd card and it still doesn't work and I don't know which one I should replace

PabloMK7, 2023-01-28:
Hello, you will get better support if you ask in the Nintendo homebrew discord server, as this page is meant for tracking bugs and not giving support: https://discord.gg/C29hYvh

closed 2023-01-28 by suppai