LumaTeam/Luma3DS #1833: [FR] per button Turbo/Autofire toggles

eadmaster, 2023-01-22:
Allow to set Turbo/Autofire induvidually for each button via the [Rosalina menu](https://github.com/LumaTeam/Luma3DS/wiki/Rosalina).

Useful for games that requires lot of button mashing like Senran Kagura.

Also asked here:
 - https://gbatemp.net/threads/turbo-mode.533901/
 - https://gbatemp.net/threads/3ds-autofire.468047/

PabloMK7, 2023-01-28:
This feature is probably better suited for [rehid](https://github.com/hax0kartik/rehid) (possible it has it already).

Golem642, 2024-05-06:
It does now, still a beta feature but it works pretty well

Golem642, 2024-05-06:
> Afik the Rehid implementation cannot be turned off while ingame

Wdym ? It's literally bound to a key combo of your choice, and you can toggle it on or off anytime