LumaTeam/Luma3DS #1835: Rosalina API ?

SeleDreams, 2023-02-05:
I was wondering if rosalina had any APIs homebrews could interact with ?
I wanted to make a custom build of the homebrew launcher that only runs in net load mode and automatically enables the "enable debugger on next app" setting of rosalina, but i am not too sure about how i could tell rosalina to do that in C/C++

SeleDreams, 2023-02-05:
looking more into the source it seems like normally it should have been possible to request it from other homebrews through the hbldr module but i don't know why it seems to have been removed

[image]

SeleDreams, 2023-02-06:
I've been learning how luma is structured and I think I start to get it, seems like Rosalina does have a notification for it that I could call from homebrew launcher
{ 0x1000,                       handleNextApplicationDebuggedByForce    },
however it seems like it lacks a notification to start the debugger so i'll probably create one and do a pull request once i get it working

SeleDreams, 2023-02-06:
I ended up understanding better how luma and its services communicate with other homebrews

I started adding some features to rosalina to allow it to communicate with the homebrew launcher when it comes to debugging homebrews

SeleDreams, 2023-02-06:
well I did try by adding this to rosalina
```
static const ServiceManagerServiceEntry services[] = {
    {"rosalina:dbg", 2, handleRosalinaDebugger, true}, {NULL}};

```

and then in homebrew launcher

```
Result r = svcConnectToPort(&rosalinaHandle, "rosalina:dbg");
```

but it doesn't seem to be working as i get the result code 0xE0181E
I can't find documentation on this

closed 2023-02-06 by SeleDreams

TuxSH, 2023-02-07:
Hi,

The reason most hb:ldr commands were removed is because I moved it from Rosalina to loader entirely (it should have been like that in the first place anyway!).

I plan to move the "debug next app" to a "monitor" GDB command like Atmosphère-NX does, it's probably much simpler that way.

SeleDreams, 2023-02-07:
i see.
honestly the main reason i was working on all this had the simple goal of easily being able to debug homebrews without going through tons of menus etc before every single session