LumaTeam/Luma3DS #1837: [request] Option to force audio routing to onboard speakers

raekuul, 2023-02-09:
As an extension to #1828, can we also have an option to force audio to route to onboard speakers (ignoring the headphone jack entirely)?  My unit is water damaged in such a way that the headphones report that they're always "in use."  In 3DS Mode this means I need to plug in headphones to get sound.

I know that the onboard speakers themselves still work, though:
1.  The Camera App still correctly plays the "taking a picture" sound through the onboard speakers
2.  The version of OPEN_AGB_FIRM that I have installed doesn't properly detect headphones and so plays through the onboard speakers anyway.

(originally requested in the comments to #752 and I don't know if this was kept since #752 was closed as unmaintainable shortly thereafter)

TuxSH, 2023-02-10:
Done, here's a build:

[boot.zip](https://github.com/LumaTeam/Luma3DS/files/10702796/boot.zip)

closed 2023-02-10 by TuxSH

raekuul, 2023-02-10:
Works for 3DS mode, thank you

kuyukurt2923, 2023-02-25:
this seems to be not working on my regular 3ds ll. Speaker works on camera as well but when I saved the settings on my luma menu with the force audio X out on the speakers, nothing happens

kuyukurt2923, 2023-02-26:
Oh and headphones does not work for me as well.  No sound on speakers and headset

123321mario, 2023-03-06:
this actually revived my new 3ds speakers but can't launch ctgp anymore
could this be pushed to the main release?

Zarathestrio, 2023-04-26:
This fix almost completely works for me, however I've found that certain applications still do not play from my speakers. Most specifically, it seems to be any applications that can't be minimized, such as DS games or GBA games. If anyone else has had this issue, any help would be greatly appreciated!

Masamune3210, 2023-04-26:
TWL_FIRM and AGB_FIRM are separate from anything Luma touches. It would take modifying or replacing those firms for this to work for DS or GBA games.

JustCoddy, 2023-08-01:
> Done, here's a build:
> 
> [boot.zip](https://github.com/LumaTeam/Luma3DS/files/10702796/boot.zip)

hey so i’m kinda new to this and i have the same problem i just don’t know where to put this file or how to boot it i need help

commit ef1072f9 (2023-02-10): Add option to force audio routing to speakers, closes #1837

Also refactor ini file a little bit