LumaTeam/Luma3DS #1841: Wumiibo Menu no longer opening since Luma 12.0

InsanePrimate, 2023-02-11:
System model: [New 2DS XL]

SysNAND version: [11.16.0-48E]

Entrypoint: [Fastboot v1.2]

Luma3DS version: [v12.0]

Luma3DS configuration/options:

Screen brightness: (4 )

Splash: ( Off)

PIN lock: ( Off)

New 3DS CPU: ( Off)

Homebrew autoboot (Off)

--

Enable loading external FIRMs and modules: (X)

Enable game patching: (X)

Redirect app. syscore threads to core2 ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

Explanation of the issue:

Wumiibo Menu is no longer opening since Luma 12.0

Steps to reproduce:

1. Install as detailed here https://github.com/hax0kartik/wumiibo

2. Open game and reach the screen where it says to place the amiibo.

3. Press L + DOWN + START to bring up Wumiibo Menu and select the amiibo (won't open since 12.0)

urherenow, 2023-02-11:
This isn’t the wumiibo github

InsanePrimate, 2023-02-11:
> This isn’t the wumiibo github

I assumed it was worth mentioning that the new build had an unintended effect. My apologies! So whenever a new Luma version breaks some other Homebrew that worked perfectly fine before, you are not interested in knowing about it. Noted! I will refrain from posting such issues in the future. May I suggest adding this to the "before you post - list" so that others do not waste their time on reporting issues you are not interested in knowing about?

urherenow, 2023-02-11:
I’m not the owner here. But when a specific homebrew breaks because of a firmware or CFW update, it’s often a case of said homebrew needing an update, or simply needing to be built with the latest libraries. You should always check with that homebrew creator first. They will contact the CFW author(s) if necessary. 

Here, you as a user, should report problems with Luma itself, not problems with other projects.

cooolgamer, 2023-02-12:
That issue can be resolved by enabling both "enable external firms and modules" and "enable game patching" in luma config, that's a new thing in luma v12.0 to allow process patches (like wumiibo)

InsanePrimate, 2023-02-12:
Both game patching and loading external firms and module options were enabled and the menu still doesn't open (tested in Metroid Samus Returns). I'll wait for a fix on hax0kartik's end.

InsanePrimate, 2023-02-14:
All right. A fix has been found! For all Fastboot users: It needs to be configured to boot Luma from the SD and NOT from CTRNAND (wasn't a problem before, but now it apparently is).

profi200, 2023-02-14:
It has been like this since forever? Luma will load the config and other stuff from CTRNAND if booted from there.

TuxSH, 2023-02-24:
The reason it needed code.bin + exheader.bin is because I never implemented CXI loading. Now that I've done that, he should distribute the cxi file instead

closed 2023-02-24 by TuxSH