LumaTeam/Luma3DS #1843: 2DS doesn't boot

r4distiso, 2023-02-17:
I have the boot.firm and boot.3dsx files on the root of my SD card but my 2DS refuses to boot. Help!

salvasin, 2023-02-19:
im having the same issue .have you found a solution?

urherenow, 2023-02-20:
This is not a support forum, and neither of you have provided any useful information whatsoever. Ask on Discord or gbatemp.

rdistiso, 2023-02-20:
> This is not a support forum, and neither of you have provided any useful information whatsoever. Ask on Discord or gbatemp.

I have access to neither as I am underage

rdistiso, 2023-02-20:
> im having the same issue .have you found a solution?

Does your 2DS give a White light for the top LED and Blue for Power LED?

Shadow322, 2023-02-20:
@urherenow can u ban them please

rdistiso, 2023-02-20:
I have diagnosed the problem just dont know how to fix it, The 2DS can't find the boot.firm

Masamune3210, 2023-02-20:
Once again, this is not a support forum. Please go to somewhere else to figure this out.

closed 2023-03-22 by TuxSH