LumaTeam/Luma3DS #1844: [reqj

frogdude66, 2023-02-24:
In a future update, could you move the 'Start InputRedirection' from 'Miscellaneous options' to the main 'Rosalina menu' so it can be more easily accessed. Thanks in advance

_Originally posted by @frogdude66 in https://github.com/LumaTeam/Luma3DS/issues/752#issuecomment-1444308329_

lividhen, 2023-02-24:
This would not be ideal for everyone as a lot of people don't use it and it would cause clutter. You can however create a fork and for the most part copy paste the option onto the main menu.

frogdude66, 2023-02-24:
How would this be possible? I have no coding knowledge whatsoever and this seems too complicated for me. Is there a guide to doing this?

lividhen, 2023-02-24:
I moved `{ "Start InputRedirection", METHOD, .method = &MiscellaneousMenu_InputRedirection },` from line 71 in `/sysmodules/rosalina/source/menus/miscellaneous.c` to line 48 in `/sysmodules/rosalina/source/menus.c`. Seems to work pretty good.
[boot.firm.zip](https://github.com/LumaTeam/Luma3DS/files/10829109/boot.firm.zip)

frogdude66, 2023-02-24:
Wow Thank you. That is very helpful

closed 2023-02-24 by frogdude66