LumaTeam/Luma3DS #1845: [Feature request] Autoboot a title/game through a button combination

iCRON862, 2023-02-25:
If autoboot the hbmenu was possible, can you make a feature to autoboot a title/game through button combinations specified in config.ini as example? Or can i modify the config.ini to make luma3ds autoboot the title without the hbmenu take over? Thx

deco1212, 2023-09-25:
That is avaliable through Fastboot3DS

mariomadproductions, 2024-03-06:
> That is avaliable through Fastboot3DS

How?

urherenow, 2024-03-06:
please learn to read. And don't start discussing other projects in the wrong github.

"You may also want to set up the other boot slots and assign key combos to them. Keep in mind you need one autoboot slot (= a slot with no key combo assigned). If you want to access the fastboot3DS menu at a later point in time, hold the HOME button when powering on the console."

Things just above and continuing in that paragraph, tell you how.

urherenow, 2024-03-06:
You're right. My bad. But it's the most rediculous thing I ever heard. Do you understand what a firm does? It sets up all of the hardware. You can't even boot the home menu without a firm, so the answer is an absolutely eye-rolling NO. You CANNOT directly boot a file without booting a firm.

mariomadproductions, 2024-03-06:
I understand. Sorry to everyone for clogging up the thread.