LumaTeam/Luma3DS #1847: Can't Access Rosalina my L button doesn't work om my 3DS

Jasmine5654, 2023-03-01:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

urherenow, 2023-03-01:
Seriously, what do you think you’re doing? YOUR hardware is broken, you KNOW this, and you’re going to submit an issue?

This space is for collaboration and reporting bugs. It isn’t a help forum.

Jasmine5654, 2023-03-01:
If you are not going to offer anything helpful then kindly stay off my post thanks. I saw another post like mine but none of the links worked for my 3ds so I need some insight.

Nemris, 2023-03-01:
@urherenow has a point - this space is an issue tracker to report bugs.

That said - how do you expect to get help without even providing your Luma3DS version? The config file format changed recently enough, after all, and we can't really know what version you're running if you don't mention it.

Tom-NeverNull, 2023-03-14:
Noob here.
Possible suggestion.

clone luma3ds
edit  Luma3DS/sysmodules/rosalina/include/menu.h 
Line 36 `#define DEFAULT_MENU_COMBO      (KEY_L | KEY_DDOWN | KEY_SELECT)` change KEY_L to a different usable key. KEY_R for example
Then compile and install?

Tom-NeverNull, 2023-03-14:
Your can change the Rosalina Key Combo by going to your SD card > Luma > config.ini and changing `rosalina_menu_combo = L+Down+Select`
for example: 
`rosalina_menu_combo = R+Down+Select`

closed 2023-03-22 by TuxSH