LumaTeam/Luma3DS #1848: closed-lid powering up boots to white screen (v12.0 stable)

al797, 2023-03-07:
**Steps to reproduce on a N2DS:**

1. power up the console without opening the lid.

2. wait a couple seconds then open the lid.

this is not an issue on v11.0.

al797, 2023-03-08:
turned out my n2ds wasnt on latest firmware. solved.

Shadow322, 2023-03-08:
> turned out my n2ds wasnt on latest firmware. solved.

It happens on latest fw still

al797, 2023-03-08:
> > turned out my n2ds wasnt on latest firmware. solved.
> 
> It happens on latest fw still

you're right lol. I guess it's not solved then.

lifehackerhansol, 2023-03-22:
Reproducible on N3DSXL as well.

lifehackerhansol, 2023-03-22:
After some discussion in a separate Discord server this is also reproducible on O3DS, which seems to suggest that this is an issue with all devices.

al797, 2023-04-16:
still an issue on 12.0.1

TuxSH, 2024-05-01:
Fixed on v13.0

closed 2024-05-01 by TuxSH