LumaTeam/Luma3DS #1849: Xenoblade EUR (Cartridge) constant crashing

mariochristo03, 2023-03-08:
**System model:**
Occurs on New 3DS, New 2DS and New 3DS XL
[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.16.0-49E

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap

**Luma3DS version:**
V12 Stable
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

For this option ^ I have tried a variation making the same results every time.
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
First of all, here is a link with a screenshot.
https://imgur.com/a/CRWvMEM
So the issue is that Xenoblade 3D constantly crashes during gameplay, and in specific spots too. prior to modding my consoles, I was able to complete this game with the same cartridge. On all consoles the latest update for the game is applied.
All my consoles are using the Sandisk red and grey micro sd cards. Two of them have a 32gb micro sd card at 32 kb cluster size, while the New 3DS has a 64gb micro sd card at 64kb cluster size. I can only think at this point that the errors lead back to Luma, as I was never getting these issues before. Any help will be greatly appreciated

**Steps to reproduce:**
There are specific variations of the crash (this has been found under a new save file.

1. In the overworld for the first town (colony 9) walking around sometimes the game simply shuts off and sends me to the 3DS home menu without saying anything went wrong, just a regular looking 3DS boot.

2. In an early cutscene where Shulk has his first set of visions holding the Mondado, the game will always crash at the exact same point on all models during a transition scene.

3. Sometimes the game's intro sequence has the tendency to crash as well

**Dump file:**

I don't have the disable arm11 option, but the error txt file was in the luma folder. If there is any way to get this dmp file please let me know.
The text in the error file is as follows:

Reported on:      2023-03-07 13:38:04
System booted on: 2023-03-07 13:36:34

Error type:       generic

Process ID:       39
Process name:     smallfol
Process title ID: 000400000f700200

Address:          0x003be910
Error code:       0xd900458b

-------------------------------------

Reported on:      2023-03-07 13:59:36
System booted on: 2023-03-07 13:38:06

Error type:       generic

Process ID:       41
Process name:     smallfol
Process title ID: 000400000f700200

Address:          0x003be910
Error code:       0xd900458b

-------------------------------------
Again, thank you for the help.

TuxSH, 2023-03-22:
0xd900458b means "hash check failed". I think you cartridge might be faulty, does this happen with a digital version of the game?

mariochristo03, 2023-03-22:
> 0xd900458b means "hash check failed". I think you cartridge might be faulty, does this happen with a digital version of the game?

Thank you for the reply, I am not home for a little bit but once I get home I will make sure to check and get back to you.

profi200, 2023-03-23:
You can try this and report back to the dev. He is looking to collect data on faulty gamecards: https://github.com/skawo/GodMode9-with-Cartridge-Fixer

mariochristo03, 2023-03-23:
So a bit of a weird update, tried the new godmode9 cart fixer. Sat 3 hours waiting because xenoblade is a larger game, booted it up and got the same issue as before. I decided I'd try some old fashioned alcohol which I rubbed on the contacts and by some miracle it made it work again. I guess my issue is solved, but it was a weird phenomenon none the less with how it would always crash in the same spot as well as fully soft resetting the entire console. @TuxSH @profi200

profi200, 2023-03-23:
That's weird. Usually with dirty contacts it will fail to initialize so it should not even appear on HOME menu. Good that it's fixed now.

closed 2023-07-16 by PabloMK7