LumaTeam/Luma3DS #1850: [SUGGESTION] Add support for animated/moving GIF splash screens

Minionguyjpro, 2023-03-09:
Please make it possible to add support for animated/moving splash screens from a GIF file, if possible. I guess it should be possible, as in the Luma3DS configuration things can also move when scrolling through (might be completely different, but should make sense why it should be possible). Because now I want to use a gif that has a fake loading animation, which should move, but it does not work...

Masamune3210, 2023-03-09:
Early boot is very limited, just because a simple config menu can react to presses, that doesn't mean that jumping to gif playback being able to be done should be next thought

Minionguyjpro, 2023-03-10:
Yes, but it might be possible, or in a rate that is low, like 1-5 frames each second or so...

PabloMK7, 2023-07-16:
Duplicate of #459

closed 2023-07-16 by PabloMK7