LumaTeam/Luma3DS #1852: Closing the Rosalina menu via an input redirected controller causes a freeze

phyr0n, 2023-03-10:
**System model:**

Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.16.0-49E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v12.0 stable

**Luma3DS configuration/options:**

Screen brightness: 4 

Splash: Off

PIN lock: Off

Homebrew autoboot: Off

--

Enable loading external FIRMs and modules: False

Enable game patching: False

Show NAND or user string in System Settings: True

Show GBA boot screen in patched AGB_FIRM: False

Force routing audio output to headphones: False

--

**Explanation of the issue:**

I am using [this input redirection client](https://github.com/TuxSH/InputRedirectionClient-Qt) and a bog standard Xbox controller. After initialising input redirection, exiting the Rosalina menu using my Xbox controller freezes the 3DS and everything becomes unresponsive. A hard restart is required at this point (holding the power button down for 10 seconds). The only (currently known) way to circumvent this is by pressing B on the 3DS to exit the Rosalina menu. 

**Steps to reproduce:**

1. Enter the Rosalina menu (L + SELECT + DOWN).
2. Enter the Miscellaneous options menu and select "Start InputRedirection".
3. Plug an Xbox controller into your PC or laptop and connect to the 3DS via InputRedirectionClient-Qt.
4. Press B on the controller to go back to the Rosalina menu.
5. Press B on the controller to (attempt to) exit the Rosalina menu. This is the point where the 3DS becomes unresponsive.

nhobson99, 2024-05-10:
This also happens on my o3DS

nonsensicalTrickster, 2024-08-04:
This is also happening to me on my o3DS XL

Constrat, 2024-08-09:
Issue persists on `v13.1.2`

M1ster000, 2024-12-20:
Same on o3DS