LumaTeam/Luma3DS #1853: Arm 11 (core 0) exception error when launching Cyanogen3DS

2020loryx, 2023-03-11:
**System model:**

New 3ds

**SysNAND version (+emu/redNAND version if applicable):**

11.16.0-49E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

v12.0-3gxldr

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Before)

Splash duration: (3s default)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

When launching Cyanogen3DS the console has an error: 
Processor: Arm11 (core 0)
Exception type: data abort
Fault status: Translation - Selection
Current process: 3dsx_app (000400000D921E00) 

**Steps to reproduce:**

1.Donload Cyanogen3DS at this link: https://www.mediafire.com/file/mrugpfw1isczwt6/Cyanogen3DS_Alpha_1.0.zip/file

2.Copy 'Cyanogen3DS' folder to '3ds' folder

3.Launch the homebrew launcher and start Cyanogen3DS

4.Now you can see 'An exception occurred'

Shadow322, 2023-03-11:
Well no longer supported outdated homebrew from 2015 the Luma3DS devs arnt going to bother fixing it

2020loryx, 2023-03-13:
😥😢

PabloMK7, 2023-07-16:
Closing as unrelated to Luma3DS

closed 2023-07-16 by PabloMK7