LumaTeam/Luma3DS #1857: 3DS update with v12 Luma and 11.16.0-49U?

UZAMBIE, 2023-03-23:
Hi. For some reason my 3DS keeps prompting me to update the system, even though it’s at it’s most recent version. This only showed up after I’d updated and modded the console, and have been using it for about a week.

I can’t use eshop, streetpass or update any games, and I don’t think I can use most licensed apps either (The apps that come with the 3DS, but those being the ones I need the most).

I’m very wary about updating it, as it seems sketchy and *especially* with the only information I’ve seen being about Luma v11 in regards to the update from September (which I have). 
My 3DS is entirely modded and I don’t have much knowledge on how to help myself, only knowing that updating could brick my 3DS or wipe the data off my console.

Can I update it safely? If not, is it some sort of virus? How do I get rid of it, and how do I get rid of the update pop-ups? If I can update it, do I remove the modded SD card and update it, or do I leave it in.

Thank you so much for reading and hopefully answering

Masamune3210, 2023-03-24:
make sure you dont have any system modules or anything like that on the sd card, if you want to be sure, the easiest way is to back up everything on your sd card to somewhere else, delete everything other than the Nintendo 3ds folder, and then copy a fresh copy of the latest release of luma and see if the issue persists

with a bootrom exploit, it doesn't matter if you update or not. you arent going to brick your system or wipe your data just by accepting a firmware update.

Yes, you can update safely, viruses arent a thing for the 3ds or consoles in general for the most part.

What is the exact wording of the message that keeps popping up?

UZAMBIE, 2023-03-25:
> make sure you dont have any system modules or anything like that on the sd card, if you want to be sure, the easiest way is to back up everything on your sd card to somewhere else, delete everything [...]

Hi, thank you so much for responding. I’ll see if I can find anything weird on my SD card tomorrow, how do I identify a system module? Deleting all the contents except the 3DS folder would leave the games anyway since they’re loaded on (…right?), so I’ll definitely take up that suggestion.

I don’t completely know if I have a bootrom exploit? I think I do, as I followed all of the instructions on the official guide, but just to be safe how can I check?

The message that I need a system update shows up every time I leave a game, or open most apps. When I open eShop, it says “A system update is available! To update, select System Update in System Settings. Come back after you’ve updated!”. When I try to update a game, it shows error code 009-1000, saying a system update is required.

urherenow, 2023-03-25:
Take it to discord retronx or gbatemp. This isn’t a help forum.

closed 2023-03-27 by TuxSH