LumaTeam/Luma3DS #1859: Bottom screen flickering after booting to Luma Config

InsanePrimate, 2023-03-29:
[New 3DS - Dual IPS]

**Sysrans version (+emu/redNAND version if applicable):**

[11.16.0-49E SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[fastboot3DS 1.2]

**Luma3DS version:**

[v12.1]

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

Enable loading external FIRMs and modules: (X)

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Bottom screen (IPS) flickers and shows ghosting of the Luma config menu after using it. Fixed after reboot. Also, already described here https://gbatemp.net/threads/bottom-screen-flickering-after-booting-to-luma-config.514309/

**Steps to reproduce:**

1. Boot into Luma config (on a dual IPS system?)

2. Press start to continue to boot into Luma. Flickering and ghosting of the Luma config Text is then present on the bottom screen. Disappears after a reboot.

StikoNiko, 2023-07-24:
I know the problem.
I have a feeling they are on large memory cards.
I don't have that problem with 128GB.
With 256 GB I have the problem when I use a start screen.
With 512GB I have it permanently.

Suggested solution:
Use 128GB card.
With 256 GB, use without a start image, password entry works.
With 512 GB I have no solution other than formatting the card to 256GB.

profi200, 2023-07-24:
It's not related to SD card capacity at all. It's a race condition/bad timing on screen init and IPS panels happen to not like bad timings.

rainlynd, 2023-07-26:
How to fix the screen init? I also happens to have the same problem.

StikoNiko, 2023-07-26:
I had this problem on all 3 Dual IPS N3DSXL.
I have this N3DSXL 3 times for me , my wife and my child.

After formating my 512GB SD Card to 256GB SD Card, 
the flickering goes away.

Using GUIFORMAT.EXE with 32768 Allocation unit size
Using another Allocation unit size, and your flickering is back.

I know, profi200 write, thats not my problem.
But my 3 devices does that not know.

With Startlogo, i have flickering too.
Password option is working without flickering.

profi200, 2023-07-26:
We have another issue for this: #1396
If i recall correctly TuxSH tried this but reverted the changes because of issues. Seems to be working fine for GodMode9 and other projects though.

PabloMK7, 2023-07-26:
Closing as duplicate of #1396 , so everything is in the right place.

closed 2023-07-26 by PabloMK7