LumaTeam/Luma3DS #1862: Ability to launch a 3dsx homebrew in extended memory mode

SeleDreams, 2023-03-31:
Hi,
I've been working on porting godot 4 to the 3DS and noticed that the 64mb allocated to the O3DS by default are a bit too short, so i tested first modifying luma to make it always start homebrews in extended memory mode and it seemed to work fine

i however of course feel like running all homebrews in extended memory mode would be overkill, that's why I'm not sure about the best way to go about it.
i tried instead modifying the homebrew launcher thinking i could make it change the value of OS_KernelConfig->app_memtype before starting my homebrew but it seems like this is actually a read only value 

maybe adding an IPC command that is called from homebrew launcher would be the best, i'm not too good at IPC checking though due to its use of bitmasking
what would you recommend ?

SeleDreams, 2023-03-31:
if possible the best would maybe even be an ability to change the ram limit at runtime but i guess that might be too difficult

PabloMK7, 2023-04-03:
I don't think it's possible to change the memory mode once the homebrew menu has been launched. Keep in mind that all homebrew apps run under the same process, and changing the memory mode requires a firmlaunch/system reboot. I can think of two solutions:

1. Add some metadata to homebrew to trigger a firmlaunch and have the homebrew menu or luma autolaunch the 3dsx that was meant to be launched.
2. Have a separate homebrew menu that runs in mode3, and allow setting different TIDs for different modes.

In my plugin loader fork I've added an IPC call to loader to override the memory mode of the application being launched, maybe that will be useful for implementing this.