LumaTeam/Luma3DS #1864: The console turns on but the blue led does not

TheHyperZtar, 2023-04-04:
**System model:**
New 2DS XL
[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.16.0-49U
[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
12.1
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (Before)

Splash duration: ( )

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: (x)

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
The console turns on perfectly but the blue led is not displayed, it only appears red when the battery is low
I already tried to use Rosalina's menu options but no result

**Steps to reproduce:**

1.Turn on the console

2.

**Dump file:**

spudpiggy, 2023-06-06:
You're definitely talking about the first led that's on the bottom? The one right above power?
You say you've already checked but just to be sure, open the Rosalina menu, then go to `System configuration... -> Toggle LEDs` and hit A.
Give it a couple seconds, it shouldn't take that long but just to be sure.
The wireless LED should also be affected _if and only if_ you have wireless on.
Close your N2DSXL and the power/wireless LEDs should come back.
If part of this didn't happen as described, try it again.

Tbh i think your power led is partially broken and can't display blue for whatever reason, just try this first.
Also who cares if your power led is broken? 0 functionality lost, you said it still goes red so you can still check if battery is low

PabloMK7, 2023-07-16:
This is probably a hardware fault, not related to Luma3DS.

closed 2023-07-16 by PabloMK7