LumaTeam/Luma3DS #1866: Luma Crashes when attempting to load AVGN Adventures, RPG Maker Fes

vexthewolf2007, 2023-04-04:
**System model:**

OLD 2DS

**SysNAND version (+emu/redNAND version if applicable):**

SYS 11.16.0-49U

**Entrypoint (How/what you're using to boot Luma3DS):**

Fastboot3DS

**Luma3DS version:**

12.0.1 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4)

Splash: (Before )

Splash duration: (5 Seconds )

PIN lock: ()

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**explination**

Installed AVGN Adventures, RPG maker fes and both crash upon loading, all other apps work as intended.

**Steps to reproduce:**

1. Install The cias via. FBI

2. Launch the games on the home menu to trigger the crash

**Dump file:**

[crash_dump_00000008.zip](https://github.com/LumaTeam/Luma3DS/files/11152897/crash_dump_00000008.zip)

vexthewolf2007, 2023-04-04:
Forgot to mention the games were installed via Hshop (Not the app)

AVGN: https://hshop.erista.me/t/19943

RPG Maker: https://hshop.erista.me/t/20314

PabloMK7, 2023-07-16:
The problem reported here seems to be related to piracy. Since piracy is an illegal activity we are unable to give any support.

closed 2023-07-16 by PabloMK7