LumaTeam/Luma3DS #1867: [Feature request] Tell the user (and/or a server) if the ROM is unknown

mariomadproductions, 2023-04-04:
I think it would be useful for 3DS game preservation if Luma3DS checked the ID and version of installed titles/inserted cartridges against a list of known ones, and then told the user or maybe sent it as (optional) telemetry to a server. That way undocumented titles/title versions/cartridge revisions could be discovered. Not all digital titles or cartridges are preserved/documented yet (despite that maybe being a common belief). To make sure homebrew/mods aren't falsely flagged up, tmd/header signatures should be checked.

profi200, 2023-04-05:
Source? As far as i know pretty much everything has been preserved/dumped. Besides that it would be out of scope for Luma and i personally would not want it to phone home without asking me.

mariomadproductions, 2023-04-05:
> Source? As far as i know pretty much everything has been preserved/dumped. Besides that it would be out of scope for Luma and i personally would not want it to phone home without asking me.

I'm part of the No-Intro team and have worked with hshop too. See the No-Intro undumped lists, and also the fact that titles can be removed from the CDN/eshop without leaving much trace (if any).

Yeah it should probably ask the user before sending anything.

Its possibly out of scope, but it would be good for preservation, as most people don't check/don't know if they have anything that isn't preserved.

profi200, 2023-04-05:
I see one big problem with this. If you ask the user to dump their title/gamecard it would make Luma even more gray area than it already is. The best you can realistically do is make users submit infos like hash, revision/version etc. I think this would fit better for GodMode9.

PabloMK7, 2023-07-16:
Hello,
this feature seems to be out of the scope of Luma3DS. It can probably be achieved by other less intrusive means such as a custom system module or homebrew application. Keep in mind that features that are only beneficial to a very small group of people cannot be implemented in a custom firmware designed for everyone, as implementing too many of them could bloat the source code.

I understand the needs for preservation, but all the code needed for this would waste CPU and RAM resources that are already sparse.

closed 2023-07-16 by PabloMK7