LumaTeam/Luma3DS #1871: [Feature Request] Add force_audio_output setting to Rosalina Menu

AmbrosiaEgg, 2023-04-08:
While the hardware issue I experience likely isn't common, I have an issue where my N3DSXL's headphone port is stuck thinking something is plugged into it even when there isn't. The recent addition of the force_audio_output = speakers option has finally allowed me to use my3DS without headphones again, but being able to change this setting more easily would be very much appreciated. If it were added to the Rosalina Menu, the hardware issue, uncommon as it may be would be reduced to a minor annoyance.

AmbrosiaEgg, 2023-04-08:
Whoops saw the old compilation thread for Rosalina feature suggestions, thought to move this suggestion there, and then realized that compilation thread is closed.

664710775, 2025-04-25:
> While the hardware issue I experience likely isn't common, I have an issue where my N3DSXL's headphone port is stuck thinking something is plugged into it even when there isn't. The recent addition of the force_audio_output = speakers option has finally allowed me to use my3DS without headphones again, but being able to change this setting more easily would be very much appreciated. If it were added to the Rosalina Menu, the hardware issue, uncommon as it may be would be reduced to a minor annoyance.虽然我遇到的硬件问题可能并不常见，但我遇到了一个问题：我的 N3DSXL 耳机接口卡住了，误以为插了东西，但实际上并没有。最近添加了 force_audio_output = speaker 选项，终于让我可以不用耳机使用我的 3DS 了。但如果能更轻松地更改此设置就更好了。如果 Rosalina 菜单里也添加了这个选项，那么这个硬件问题虽然不常见，但也会变成小麻烦。

May I ask how to call this function? I also encountered the same problem.

664710775, 2025-04-25:
> While the hardware issue I experience likely isn't common, I have an issue where my N3DSXL's headphone port is stuck thinking something is plugged into it even when there isn't. The recent addition [...]

May I ask how to call this function? I also encountered the same problem.

alvv-z, 2026-09-05:
I'm guessing this will be added by what is mentioned in https://github.com/LumaTeam/Luma3DS/issues/2231#issuecomment-4180719508

closed 2026-10-27 by TuxSH