LumaTeam/Luma3DS #1872: Arm11 (Core 0)

M1dn1ghtStr1ker, 2023-04-09:
**System model:**

[Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[12.0.1]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (off )

Splash duration: ( )

PIN lock: (off )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
I go to patch Pokemon Y with a LayeredFS file I follow all the steps provided and its gives me an error labeled "An exception has occured Arm11 (Core 0)" It also tells me the exception type is data abort, fault status - Permisson - page, current process kujira-2 (0004000000055E00) 

**Steps to reproduce:**

1.

2.

**Dump file:**

[image]

M1dn1ghtStr1ker, 2023-04-09:
Wouldn't let me zip for some reason  so I screen shot it.

Masamune3210, 2023-04-09:
please upload the dump, sending a screenshot of a binary file being opened in notepad is doubly useless. Just zip the file or change the extension to one of the supported upload formats

Yotlan, 2023-06-01:
I have the same error with the last update of Luma3DS and my old 3DS (1.17). 
[crash_dump_00000010.zip](https://github.com/LumaTeam/Luma3DS/files/11627383/crash_dump_00000010.zip)

Yotlan, 2023-06-01:
Processor: Arm11 (core 0)
Exception type: data abort
Fault status: Translation - Section
Current process: dish (00040000001c1d00)

Register dump:

r0             ffffffff            r1             00000000
r2             ffffffff            r3             00000001
r4             0ffff9f0            r5             00000000
r6             00000000            r7             00000014
r8             00000100            r9             00000010
r10            00000015            r11            00502d08
r12            00000000            sp             0ffff998
lr             001003ed            pc             00100124

cpsr           00000030            dfsr           00000005
ifsr           00001009            far            00000014
fpexc          40000700            fpinst         eeb40ac8
fpinst2        eeb40ac8
FAR            00000014            Access type: Read

Code dump:

001000f6:  03 c8 35 00 00 90 01 91  08 e0 79 5d 6a 46 02 a8   |..5.......y]jF..|
00100106:  de f1 3b fd 01 28 00 d1  76 1c 6d 1c 20 78 80 06   |..;..(..v.m. x..|
00100116:  02 d5 e0 69 b0 42 05 dd  06 98 85 42 ed db 78 5d   |...i.B.....B..x]|

Stack dump:

0ffff998:  00 00 00 00 00 00 00 00  d8 f9 ff 0f ad ec 2d 00   |..............-.|
0ffff9a8:  f0 f9 ff 0f 14 00 00 00  ff ff ff ff f0 f9 ff 0f   |................|
0ffff9b8:  73 00 00 00 68 fa ff 0f  00 00 00 00 ed 03 10 00   |s...h...........|
0ffff9c8:  f0 f9 ff 0f 1b 03 10 00  5c 4d 3e 33 f0 f9 ff 0f   |........\M>3....|
0ffff9d8:  68 fa ff 0f 00 01 00 00  68 fa ff 0f 70 fc ff 0f   |h.......h...p...|
0ffff9e8:  70 fc ff 0f 37 04 10 00  00 00 00 00 41 04 10 00   |p...7.......A...|
0ffff9f8:  30 fa ff 0f 11 04 10 00  ca fc ff 0f 01 00 00 00   |0...............|
0ffffa08:  00 00 00 00 00 00 00 00  2b 00 00 00 30 34 f4 30   |........+...04.0|
0ffffa18:  41 63 68 69 65 76 65 49  6e 66 6f 5f 30 31 37 00   |AchieveInfo_017.|
0ffffa28:  68 00 00 00 1f 63 2b 00  c6 fa ff 0f 6e fc ff 0f   |h....c+.....n...|
0ffffa38:  d8 fc ff 0f 70 fa ff 0f  fe ff 00 00 18 a2 24 00   |....p.........$.|
0ffffa48:  2c bb 41 33 20 c0 41 33  00 01 00 00 70 fa ff 0f   |,.A3 .A3....p...|
0ffffa58:  d0 be 40 00 84 9f 24 00  70 fa ff 0f 70 fe ff 0f   |..@...$.p...p...|
0ffffa68:  14 00 00 00 68 00 00 00  56 00 6f 00 75 00 73 00   |....h...V.o.u.s.|
0ffffa78:  20 00 61 00 76 00 65 00  7a 00 20 00 72 00 65 00   | .a.v.e.z. .r.e.|
0ffffa88:  6d 00 70 00 6f 00 72 00  74 00 e9 00 20 00 6c 00   |m.p.o.r.t... .l.|
0ffffa98:  65 00 20 00 62 00 6f 00  6e 00 75 00 73 00 0a 00   |e. .b.o.n.u.s...|
0ffffaa8:  43 00 65 00 69 00 6e 00  74 00 75 00 72 00 65 00   |C.e.i.n.t.u.r.e.|
0ffffab8:  20 00 6e 00 6f 00 69 00  72 00 65 00 20 00 00 00   | .n.o.i.r.e. ...|
0ffffac8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffad8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffae8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffaf8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb08:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb18:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb28:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb38:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb48:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb58:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb68:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb78:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb88:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffb98:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffba8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffbb8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffbc8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffbd8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffbe8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffbf8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0ffffc08:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
[... 40 lines cut ...]
0ffffe98:  01 00 00 00 f4 7f 3c 33  00 00 00 00 00 00 00 00   |......<3........|
0ffffea8:  96 56 00 00 38 2f 25 00  2c bb 41 33 48 ff ff 0f   |.V..8/%.,.A3H...|
0ffffeb8:  3c 8b 3c 33 98 90 24 00  00 00 20 c3 00 00 f0 42   |<.<3..$... ....B|
0ffffec8:  6c bb 41 33 00 00 00 00  00 00 00 00 00 00 00 00   |l.A3............|
0ffffed8:  00 00 00 00 2c bb 41 33  48 ff ff 0f 3c 8b 3c 33   |....,.A3H...<.<3|
0ffffee8:  2c bb 41 33 01 00 00 00  f4 7f 3c 33 00 00 00 00   |,.A3......<3....|
0ffffef8:  00 00 00 00 f0 a5 38 00  00 00 e2 42 80 00 00 00   |......8....B....|
0fffff08:  00 00 00 00 00 00 00 00  80 00 00 00 00 00 00 00   |................|
0fffff18:  00 00 00 00 01 00 00 00  48 ff ff 0f 0c 95 2d 00   |........H.....-.|
0fffff28:  c8 7f 3c 33 90 ff ff 0f  84 7c 3c 33 c8 7f 3c 33   |..<3.....|<3..<3|
0fffff38:  01 00 00 00 8c 8e 01 08  00 00 00 00 20 2c 3b 00   |............ ,;.|
0fffff48:  03 00 00 00 05 00 00 00  07 00 00 00 0c 95 2d 00   |..............-.|
0fffff58:  38 23 58 00 34 16 4f 00  08 25 58 00 8c 8e 01 08   |8#X.4.O..%X.....|
[... cut ...]
0fffffa8:  84 68 10 00 f0 68 10 00  04 69 10 00 c8 7f 3c 33   |.h...h...i....<3|
0fffffb8:  18 69 10 00 4c 69 10 00  00 00 00 00 10 64 10 00   |.i..Li.......d..|
0fffffc8:  00 00 00 ff 54 64 10 00  00 00 00 ff 02 04 00 00   |....Td..........|
0fffffd8:  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00   |................|
0fffffe8:  00 00 00 00 00 00 00 00  00 00 00 00 94 4a 10 00   |.............J..|
0ffffff8:  00 00 00 00 20 00 10 00                            |.... ...|

Above is the copy-past after I parse with luma the dump file

PabloMK7, 2023-07-16:
The problem reported here is not related to Luma3DS, but it is triggering Luma3DS' exception screen, which improves on the console's "the software closed due to an error" message.

Sadly, we cannot do anything to fix the issue and you should contact the developers of the faulty modified files so they can fix it.

closed 2023-07-16 by PabloMK7