LumaTeam/Luma3DS #1873: Question how update luma3ds when move hacks to internal storage on new 3ds xl.

Riot20058, 2023-04-10:
I have question so move my hacks to internal storage so my question how is on new 3ds xl is how update my luma3ds system when a new ver comes out.

Riot20058, 2023-04-10:
Information do it come for retro game crops video guides on how to hack a 3ds.

urherenow, 2023-04-10:
Github isn’t a support forum. It’s for reporting bugs and collaboration.

https://3ds.hacks.guide/finalizing-setup.html
Section VI - CTRNAND Luma3DS

Riot20058, 2023-04-10:
@urherenow did that my question is how update without breaking the whole thing because like do it new ver of luma3ds comes out.

Riot20058, 2023-04-10:
Like this know so don't put yet outher bugs report of something goes wrong.

Riot20058, 2023-04-10:
@urherenow think do put luma3ds update files onto sd card and after words go into godmode to copy luma3ds update into the 3ds internal storage is that correct way to do it.

Riot20058, 2023-04-10:
@urherenow video saying just copy file into internal storage By fbi as way update luma3ds probably best way to do it.

closed 2023-04-10 by Riot20058

Korbaux, 2023-04-14:
dont use a video guide