LumaTeam/Luma3DS #1874: Adjust battery percentage that the console sees

weirdal3333, 2023-04-12:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.16.0-49U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap

**Luma3DS version:**

12.0.1

**Luma3DS configuration/options:**

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

Can you adjust or invert the displayed battery percentage on the console? Currently the console says it's at half battery when Rosalina shows 25%. 

**Steps to reproduce:**

I will attach a screenshot for context 
[image]

**Dump file:**

patataofcourse, 2023-04-12:
just to clarify - the battery icon doesn't display the actual battery level, in my experience it goes down to 3 bars at ~50%, 2 bars at ~25%, 1 bar / red at ~10% and 1 bar flashing at ~5%

weirdal3333, 2023-04-14:
Yes. I'd much rather that it shows 1 bar at like 25%, flashing at 10-15%... Wear and tear on the batteries could be minimized if the handhelds got plugged in sooner or you had just enough battery to save and quit!
When older batteries hit 10%, they have about 2 minutes left of battery, sometimes even only a few seconds.

PabloMK7, 2023-07-16:
Hello,
this feature seems to be out of the scope of Luma3DS. It can probably be achieved by other less intrusive means such as a custom system module or homebrew application. Keep in mind that features that are only beneficial to a very small group of people cannot be implemented in a custom firmware designed for everyone, as implementing too many of them could bloat the source code.

A patch to the home menu would do the trick in this case.

closed 2023-07-16 by PabloMK7