LumaTeam/Luma3DS #1876: [IDEA] Translation System with Multiple Languages

Minionguyjpro, 2023-04-15:
I would like to see a translation system that allows to have Luma3DS in multiple languages, such as your own one and others.

PabloMK7, 2023-07-16:
This would be an interesting feature indeed, as people won't need to recompile or modify the source to apply language patches. However it has downsides such as increased binary size because of the extra logic to handle language packs.

Minionguyjpro, 2023-07-17:
> This would be an interesting feature indeed, as people won't need to recompile or modify the source to apply language patches. However it has downsides such as increased binary size because of the extra logic to handle language packs.

Yes, but I think it would not care for most people though. Otherwise yiu can also make it an optional feature/seperate language pack.

Komas19-new, 2023-08-13:
or atleast making it easier to modify the texts, for example, in a lang.yml file

PoorPocketsMcNewHold, 2024-07-30:
> This would be an interesting feature indeed, as people won't need to recompile or modify the source to apply language patches. However it has downsides such as increased binary size because of the extra logic to handle language packs.

You are indeed talking about handling external language packs ? Packing community translation into Luma, while practical would indeed increase too much the binary size, but does loading an external language file placed in the luma folder on the SD perhaps works ?
I'm wondering if the pre-boot luma option menu has access to the SD card content already at this stage to load the translation.

Minionguyjpro, 2024-07-30:
> You are indeed talking about handling external language packs ? Packing community translation into Luma, while practical would indeed increase too much the binary size, but does loading an external [...]

Well if Luma at boot can load itself from the SD card then it can also do that I'd say. That's indeed a good solution to the binary sizes problem!