LumaTeam/Luma3DS #1881: Arm11 (core 0)

vacaw, 2023-04-18:
**System model:**

Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

I can't tell
**Entrypoint (How/what you're using to boot Luma3DS):**

My SD card
**Luma3DS version:**

12.0
**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: ( After)

Splash duration: ( )

PIN lock: ( Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
When I try to acess System Settings to configure my wi-fi, the console shows a black screen with white letters, saying in red An exception occurred. I can't upload the dmp file cause I don't have winrar to zip it in .zip, so I uploaded a photo of the error.

**Steps to reproduce:**

1.

2.

**Dump file:**

[image]

Axo1409, 2023-06-07:
same issue

Axo1409, 2023-06-17:
ok so i fixed the issue by disabling plugin loader for games/apps that this happens to

vacaw, 2023-06-19:
can you tell me how do you did this? it is using the god mode (pressing select then power button)? what plugings did you disable?

PabloMK7, 2023-07-16:
Have you ever installed a 3GX plugin? If you did you can try turning off the plugin loader from the Rosalina menu. (Press L + DPAD DOWN + Select in the home menu). If this still doesn't work, please ask for help in the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh).

closed 2023-07-16 by PabloMK7