LumaTeam/Luma3DS #1882: Autoboot crash when forcing shutdown

kynex7510, 2023-04-30:
When autoboot to 3DS is enabled, by forcing a shutdown (ie. holding the power button) NS will break on next boots. Booting with autoboot disabled solves the issue, but then debugging hostile homebrew that hang the system becomes really tedious.

TuxSH, 2023-07-07:
This is a known issue I know happens when Home Menu fails to lauch when autobooted like that, but I'm not sure what's happening.

> Is there even any other way to shut the system down after you've autobooted a 3dsx?

Short press the POWER button and wait for a little bit for the Home Menu's shutdown screen to appear.

pdpapper, 2023-07-11:
Short pressing the power button always locks the system for me if I autoboot a 3dsx. Is it really supposed to work? Autobot mem type is set to normal (0) in the ini. Trying other mem types, just results in an ARM11 exception (for 3dsx_app) at boot.

This is on a New 2ds LL, booted with fastboot3ds, and hbmenu 2.4.1 as boot.3dsx (as well as others).

I've also noticed that some 3dsx apps (DaedalusX64 especially) have graphical issues and are more pronr to crashing when I autoboot instead of going through the 3ds menu. I thought it was probably because some system services were unavailable, but maybe it's something else.

Rosalina is still available, though, I just hadn't noticed because it was still set to the default key shortcut. Shutting down through Rosalina does not interfere with the next boot.

commit cc46e16d (2024-06-12): Fix hb autoboot stuck in exception at boot after a forced shutdown

Closes #1882.

Whenever power button is held long enough ("force shutdown"), mcu sysmodule
stores a flag in free reg 0. It will clear it next boot.

During that next boot, if that flag was set and if CFG_BOOTENV.bit0 is set
(warmboot/firm chainload, i.e. not coldbooting), then main() will simulate
a "power button held" interrupt (after upgrading mcu fw if necessary -- it
will reboot console after if it has upgraded mcu fw, I guess that's one of
the reasons the flag is there). This obviously cause other processes to initiate
a  [... cut]
closed 2024-06-12 by TuxSH