LumaTeam/Luma3DS #1883: Where has the "Force audio to bluetooth" been moved to?

ILFforever, 2023-04-30:
I'm doing a bluetooth mod and I found the setting Force audio to bluetooth quite usefull but in 12.0.1 it has seemingly disappered. I've seen the patch notes saying they moved it but I can't for the life of me find it. Could anyone help?

Gruetzig, 2023-04-30:
its been moved to the config.ini file(you find that file in the luma folder), set force_audio_output in there to 1 to enable the option. It has been moved because people accidentally enabled it and then complained about having no sound lol

Masamune3210, 2023-05-01:
How do you accidentally enable a setting without reading it lmao

ILFforever, 2023-05-01:
> its been moved to the config.ini file(you find that file in the luma folder), set force_audio_output in there to 1 to enable the option. It has been moved because people accidentally enabled it and then complained about having no sound lol

I understand the notion to do that but I really would still like an option to turn it on in the device not in the ini. As that requires a pc or another device (as far as I know) or maybe I'll just stick to using 12.0.0

Gruetzig, 2023-05-01:
> How do you accidentally enable a setting without reading it lmao

Some people think they have to enable everything for some reason
> I understand the notion to do that but I really would still like an option to turn it on in the device not in the ini. As that requires a pc or another device (as far as I know) or maybe I'll just stick to using 12.0.0

Why is that? You could revert 64eda63 and 3e63994 and build it manually.
By the way, GodMode9, even though its awful, has a hexeditor you can use to set the 0 to 1 and 1 to 0.

Masamune3210, 2023-05-01:
It also has a plain text editor unless I'm misremembering

Gruetzig, 2023-05-01:
that is a viewer, you cant edit with it unless i'm misremembering, none of my 3DSes are in a state where they can launch GodMode9 so I cant check

ILFforever, 2023-05-01:
> Why is that? You could revert [64eda63](https://github.com/LumaTeam/Luma3DS/commit/64eda6313282e6719d0d35c5a03b44413fea0257) and [...]

Right I guess I'll try to built in manually. Even though I have some background in coding I have no idea where to start manually build Luma boot files. Could you please provide some documents or pointers?

urherenow, 2023-05-02:
This isn’t a help forum, and your “pointers” are inconspicuously on the front page…

There are also a number of tutorials on gbatemp.

closed 2023-07-07 by TuxSH