LumaTeam/Luma3DS #1884: Exception when using GTS in Pokemon X

Dragonpark, 2023-04-30:
**System model:**
New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.16.0-49U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
v12.0.1

**Luma3DS configuration/options:**

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

Hbmenu autoboot: (Off)

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Redirect app. syscore threads to core2 ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
When connecting attempting to use the GTS in a cartridge copy of Pokemon X , I will routinely receive an error message in red stating 'An exception occurred'. I have not yet tested a digital copy of either Pokemon X/Y as I do not have the progress in that copy of the game and havent copied my save over yet.

**Steps to reproduce:**
1. Start Pokemon X and connect to Wi-Fi
2. Start GTS and search for a pokemon to trade
3. Scroll to the second pokemon in the results and the game will crash. 
I was able to reproduce the steps above multiple times.

**Dump file:**
[crash_dump_00000000.zip](https://github.com/LumaTeam/Luma3DS/files/11361580/crash_dump_00000000.zip)

**Screenshot of error**
[image]
[image]

Dragonpark, 2023-04-30:
So there were some inconsistencies in my 3ds from when I uploaded the last dump. I found the boot.firm and boot.3dsx files on my system had widely different dates on them for some reason, so I don't think they were both updated when I last updated everything on my system. I have since re-ran through the guide on updating Luma. The RTC time on the system was also reset to 2001. This was fixed, but I can't imagine this would have any impact.

Given the above, here are some additional crash dumps when attempting to use the GTS in some of the other pokemon games. These should all be dumps from attempting to use the GTS in Pokemon X, Pokemon Y and Pokemon Alpha Sapphire.

[crash_dumps.zip](https://github.com/LumaTeam/Luma3DS/files/11361720/crash_dumps.zip)

PabloMK7, 2023-07-16:
The problem reported here seems to be related to corrupted storage, please follow the next steps:

If your game is a physical copy or a cartridge, take a look at [this post](https://github.com/skawo/GodMode9-with-Cartridge-Fixer). However, if your game is a digital copy, or if you have installed any updates or DLC to the game, your SD card may be corrupted. The following steps may help:

1) Remove the SD card from your 3DS and insert it into your computer. (Or any other device that is able to read SD cards.)

2) Copy all the contents from your SD card and paste them to a folder in your computer. This step is very important to prevent data loss.

3) Use your preferred formatting tool to format your SD card to FAT32 (even if it is already formatted to FAT32). If you need a guide read [this](https://3ds.hacks.guide/formatting-sd-(windows)).

4) Copy all the files you backed up in step 1 to your SD card again. This will rebuild the file system and remove any possible corruption.

5) Try to replicate the steps where SD corruption was causing issues. If you still are having issues, you will have to reinstall the application as the files were damaged (corruption won't damage them again). Make sure to backup the save file with Checkpoint before doing the reinstall!

6) If after doing the steps listed above you are still having issues, or corruption happens again in a short time, your SD card may be damaged or unreliable and has to be replaced. You can make sure by following [this guide](https://3ds.hacks.guide/h2testw-(windows)).

If the steps listed here don't help, please consider joining the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh).

closed 2023-07-16 by PabloMK7