LumaTeam/Luma3DS #1888: Crash by installing bades

Aircool08, 2023-05-15:
**System model:**

[New 3DS XL]

**SysNAND version (+emu/redNAND version if applicable):**

[11.16.0-49E]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[The newest from 15.5.23]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

I dont changed something there
Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

I dont changed something there
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.I started the programm with that you can install badges

2. When the programm says "Attemting to write exdata" it crashes
[crash_dump_00000003.zip](https://github.com/LumaTeam/Luma3DS/files/11477601/crash_dump_00000003.zip)

**Dump file:**

Masamune3210, 2023-05-15:
Badge Arcade works fine and anything else is homebrew, of which issues involving it should be taken up with the homebrew author.

Aircool08, 2023-05-15:
but the Errow comes from Luma3ds so im opening my Issue here. I also renamed the badges with the japanse names, but it still crashes

Korbaux, 2023-05-15:
the error luma3ds is giving is caused by said homebrew though.

Shadow322, 2023-05-15:
> the error luma3ds is giving is caused by said homebrew though.
> […](#)

Then it's not a Luma3DS problem

Masamune3210, 2023-05-16:
> but the Errow comes from Luma3ds so im opening my Issue here. I also renamed the badges with the japanse names, but it still crashes

When a program on your computer or console crashes, do you go to Microsoft or Sony or do you go to the people who made the program......

closed 2023-07-07 by TuxSH