LumaTeam/Luma3DS #1889: System Unusable???

rdistiso, 2023-05-18:
**System model:**
Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**
11.16.0-49U

**Entrypoint (How/what you're using to boot Luma3DS):**
B9S v1.4

**Luma3DS version:**
v12.0
[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: (? )

Screen brightness: (? )

Splash: (None )

Splash duration: ( )

PIN lock: ( None)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: (Y )

Show NAND or user string in System Settings: ( ?)

Show GBA boot screen in patched AGB_FIRM: (N )

Set developer UNITINFO: ( N)

Disable Arm11 exception handlers: (N )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**
Before you ask, my SD is not corrupt. I modded my friend's 2DS using Luma v12.0, after I install B9S the 2DS doesn't boot but I get a white notification LED. I can confirm that both boot.firm and boot.3dsx were both on the SD card.

**Steps to reproduce:**

1.  Mod an Old 2DS using Luma v12.0.0

2.

**Dump file:**

spudpiggy, 2023-06-09:
They have to be on the root of the SD card, not inside any folders.
Also, have you tried another SD card just in case it works? If you're using a microSD, try another adapter: maybe the one you're using right now isn't making a proper connection.

closed 2023-07-07 by TuxSH

rdistiso, 2023-10-18:
Thank you, that worked