LumaTeam/Luma3DS #1890: Input Redirection Crashing Randomly

cjcocokrisp, 2023-05-22:
**System model:**
New 3DS XL 

**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.16.0-49U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
v12.0

**Luma3DS configuration/options:**

Screen brightness: ( 4)

Splash: ( Before)

Splash duration: (Not an option )

PIN lock: (Off)

New 3DS CPU: ( Off)

--

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

--

**Explanation of the issue:**
I am currently working on a machine learning side project where I am teaching a neural network to identify shiny pokemon so I can automate shiny hunts. One of my goals is to have everything done on console and not emulator so I'm using kit kat slim to get footage to my pc and input redirection to control the 3ds. I'm using a python script that uses pyautogui to make the inputs and then a keyboard to xinput program called xoutput to make the keyboard inputs work as a controller. I was running the input script to do the shiny hunt and occasionally the 3ds will crash for whatever reason that I cannot think of. It seems to be after a soft reset when it crashes. Sometimes it will run for 4 hours others 2 hours and sometimes even only 10 minutes. I have done some googling on the issue and I found one issue. 

**Steps to reproduce:**

1. Use input redirection for a period of time

2. When Pokemon is soft reset 3ds may crash at random

**Dump file:**
attached 
[crash_dump.zip](https://github.com/LumaTeam/Luma3DS/files/11526761/crash_dump.zip)

PabloMK7, 2023-07-16:
Input redirection doesn't is not involved in any way with the ARM9 core, which is the one crashing in your case. This kind of crashes are related to SD corruption, so I'll leave this here:

---------------------------------------

The problem reported here seems to be related to corrupted storage, please follow the next steps:

If your game is a physical copy or a cartridge, take a look at [this post](https://github.com/skawo/GodMode9-with-Cartridge-Fixer). However, if your game is a digital copy, or if you have installed any updates or DLC to the game, your SD card may be corrupted. The following steps may help:

1) Remove the SD card from your 3DS and insert it into your computer. (Or any other device that is able to read SD cards.)

2) Copy all the contents from your SD card and paste them to a folder in your computer. This step is very important to prevent data loss.

3) Use your preferred formatting tool to format your SD card to FAT32 (even if it is already formatted to FAT32). If you need a guide read [this](https://3ds.hacks.guide/formatting-sd-(windows)).

4) Copy all the files you backed up in step 1 to your SD card again. This will rebuild the file system and remove any possible corruption.

5) Try to replicate the steps where SD corruption was causing issues. If you still are having issues, you will have to reinstall the application as the files were damaged (corruption won't damage them again). Make sure to backup the save file with any save manager before doing the reinstall!

6) If after doing the steps listed above you are still having issues, or corruption happens again in a short time, your SD card may be damaged or unreliable and has to be replaced. You can make sure by following [this guide](https://3ds.hacks.guide/h2testw-(windows)).

If the steps listed here don't help, please consider joining the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh).

---------------------------------------

By the way, how are you doing the soft reset? It's possible you are interrupting the accesses to the SD and that's why it is being corrupted.

closed 2023-07-16 by PabloMK7

cjcocokrisp, 2023-07-17:
For the soft reset I am hitting L, R, Start, and Select it's on a Pokemon game. I have a pyautogui script thats mapped to press those buttons and I have xoutput open so the keyboard presses are a virutal xinput device.