LumaTeam/Luma3DS #1893: hey guys i was wondering if you're going to add a patch that can bypass the latest 3ds update

tehyeetlord, 2023-05-25:
**System model:**

[New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

Splash duration: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Set developer UNITINFO: ( )

Disable Arm11 exception handlers: ( )

Enable Rosalina on SAFE_FIRM: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Masamune3210, 2023-05-25:
There is no point. You already have a exploited console. Just update if you are bothered by the nagging.

Shadow322, 2023-05-25:
> **System model:**
> 
> [New 2DS XL, New 3DS XL, New 3DS, Old 2DS, Old 3DS XL, Old 3DS]
> 
> **SysNAND version (+emu/redNAND version if applicable):**
> 
> [e.g. 11.13.0-45U SysNAND, 11.13.0-45E EmuNAND]
> 
> **Entrypoint (How/what you're using to boot Luma3DS):**
> 
> [e.g. Boot9Strap/Sighax, etc.]
> 
> **Luma3DS version:**
> 
> [e.g. v10.2 stable or if using non-releases specify the commit like this https://github.com/LumaTeam/Luma3DS/commit/0543c208fd154e6326ea5da8cbf66ffcbdef010c]
> 
> **Luma3DS configuration/options:**
> 
> Default EmuNAND: ( )
> 
> Screen brightness: ( )
> 
> Splash: ( )
> 
> Splash duration: ( )
> 
> PIN lock: ( )
> 
> New 3DS CPU: ( )
> 
> --
> 
> Autoboot EmuNAND: ( )
> 
> Use EmuNAND FIRM if booting with R: ( )
> 
> Enable loading external FIRMs and modules: ( )
> 
> Enable game patching: ( )
> 
> Show NAND or user string in System Settings: ( )
> 
> Show GBA boot screen in patched AGB_FIRM: ( )
> 
> Set developer UNITINFO: ( )
> 
> Disable Arm11 exception handlers: ( )
> 
> Enable Rosalina on SAFE_FIRM: ( )
> 
> --
> 
> **Explanation of the issue:**
> 
> **Steps to reproduce:**
> 
> 
> **Dump file:**

The patch is to update and the nag will be bypassed

Masamune3210, 2023-06-20:
Yeah...im not reading that brick wall of text lol
Does the update break exploits? Yes. 
Does that mean that a seemingly infinite list of others don't exist to use? No.
Can they do anything about Luma once its installed on a system? Not really, not in any real or permanent sense. This isn't the switch, the bootrom and early boot isn't just gonna change one day after an update.

There are only so many hours in the day and so many hands willing to whittle it away on this. Things might get fixed when and if they ever need to, but trying to push those hands to reach into the future isn't likely going to do much other than heat the air around you.

Nazosan, 2023-06-20:
> Yeah...im not reading that brick wall of text lol

I don't see the humor in ignoring someone then responding to what you did (not) read into it.

What I'm saying is there are long term potential problems.  Putting aside that another update _could_ actually cause problems someday, there is the simple fact that things such as its phoning home at all could be a problem later on as servers eventually stop supporting the 3DS at all.  Some things -- such as the WiFi settings process -- even somewhat depend on that.  Rather than it keeping at such attempts, it's better if they are bypassed entirely.  Basically what Atmosphere does on the Switch.

Masamune3210, 2023-06-20:
I don't see the point in typing a brick wall. I didn't read it, it's just really easy to guess what you said because it tends to be the same things that always get said. If you are truly worried about it, Luma is open-source, my point still stands. there are only so many people and so much time. If you are this adamant about it, feel free to work on implementing the changes

TuxSH, 2023-07-07:
We expect people to run the latest firmware version, if anything

closed 2023-07-07 by TuxSH