LumaTeam/Luma3DS #1900: Text-altering Tomodachi Life mods do not load on US version but are reported to work fine with EU version

breckstenson, 2023-06-05:
Hi! I have been trying to get [this](https://gamebanana.com/mods/427722) mod to work on my 3DS, which basically removes all gendered pronouns from the text and replaces them with "they." I am using the US version of Tomodachi Life on a US New 2DS XL which is running the latest version of Luma. I have double checked that all the files are in the right places and that patching is enabled.

From what I have heard from the developer of the mod, this is not just an issue with me. Other players with the US version Tomodachi Life are experiencing the same thing with this mod and other text-altering mods. What's weird is that players with the EU version do not have this issue, and everything works fine on Citra. The developer has suggested that because there is very little difference between the EU and US files, the issue might be with Luma, so I've decided to post here to see if anyone has any suggestions/ideas.

I did see one other post on this board that has to do with modding Tomodachi Life, but it has been closed and also involved changing a different aspect of the game.

Any thoughts are appreciated! Thank you!

breckstenson, 2023-06-05:
EDIT: To clarify, the mod doesn't crash the game or anything like that; it simply does not alter anything.

spudpiggy, 2023-06-06:
"very little difference between the EU and US files" i haven't played the US version but ikthere are way more differences between the US/UK versions of this game than between the US/UK versions of most other games (the extra languages don't count), like I'm pretty sure all the item descriptions are different.
Be 100% sure you're downloading the US version of the mod and that you're applying it correctly.

breckstenson, 2023-06-06:
Thanks for the response! When I said very little difference, I meant difference in how these specific mod files are set up. Yes, the text is different, but I believe the developer is referring to there being very little difference in the structure of how they changed them. And like I mentioned above, this issue is not just with me. Several US players have commented on the page as well as contacted the developer reporting the same thing. I will directly link to the developer's response to these comments [here](https://gamebanana.com/posts/10806811).

CZ2746isback, 2023-06-16:
It's not just the text. It's every other mod I've tried. I've tried music replacement, text replacement, model replacement, etc. and it does not work on the USA version

breckstenson, 2023-06-17:
That's a shame :( It's so weird how it's only on the US version with Luma.

spudpiggy, 2023-06-17:
Wait, are you sure the title ID folder is named correctly? I know that title IDs are different for each region of most games. Don't just use what the gamebanana page told you, find the title ID on your own. (You probably only need to change the last 8 characters.)

CZ2746isback, 2023-06-17:
I did find the title id

PabloMK7, 2023-07-16:
Hello, this seems to be out of the scope of Luma3DS, so I'm closing the issue. Would you mind moving to a forum or discord server to discuss this issue?

PabloMK7, 2023-07-16:
Reopening as this seems to be an actual issue.

GamerXYZ2, 2023-09-03:
I'd like to add this, as it seems I was misunderstood in the issue I opened about Yu-Gi-Oh: so, I discovered it was about length of the text, but it's definitely a Luma3DS issue as a much older version of Luma3DS (when it still used arm9loaderhax) worked properly with it, so it's not the file itself but an update on Luma3DS that changed something that now causes it to crash when text exceeds a certain length.