LumaTeam/Luma3DS #1901: Keep getting "DOWNLOAD FAILED" and I'm not sure what to do

M0th-gaymer, 2023-06-06:
I keep getting this message when I try to update and I don't know what to do or what went wrong. Help!
[image]

Shadow322, 2023-06-06:
> I keep getting this message when I try to update and I don't know what to do or what went wrong. Help! [image]

Update manually

impeeza, 2023-06-06:
As stated several times on previous issues, LUMA UPDATER is deprecated, NOT supported by Luma3DS and unrelated to Luma3DS, you must to update manually.

Luma3DS and Luma Updater are completely two separated products.

Please close the issue.

closed 2023-06-11 by M0th-gaymer