LumaTeam/Luma3DS #1903: [Feature Request] LayeredFS for DS Games?

shubshub, 2023-06-07:
Last night I was trying to figure out if it was possible to do LayeredFS Game Mods for DS Games
I dumped my Pokemon Platinum game, Patched it with a Randomizer and then dumped the romFS of the new .nds file to my SD card and tried to add it as a LayeredFS mod the same way as 3DS titles

Itd be cool if it was possible to play Mods of DS Games using official DS Cartridges and then I wont need to use TwiLight Menu++ to boot the modified .nds file

Masamune3210, 2023-06-07:
Not possible. Luma isnt even running when you play a DS or GBA game, as Luma only runs under NATIVE_FIRM, while DS/DSi run under TWL_FIRM and GBA runs under AGB_FIRM.

shubshub, 2023-06-07:
As an alternative
Would it be possible to Hijack the Loading of a DS Game to do the following
Have Luma dump the DS Cart to the SD and patch in a modified romfs
And then pass the modified rom to nds-bootstrap

And then the DS Cartridge would basically be used as a forwarder, First time boot would probably be slow due to dumping

spudpiggy, 2023-06-09:
That would be kinda cool, but totally pointless since twilightmenu exists.

Masamune3210, 2023-06-09:
No point in duplicating effort. Why do all of this extra stuff when you can just dump the game, modify it, and then run it using twilight/nds-bootstrap

TuxSH, 2023-07-07:
This is not possible, you should rebuild the DS rom, which is always <= 512MB anyway.

closed 2023-07-07 by TuxSH