LumaTeam/Luma3DS #1904: Arm11 (Core 0) Errors

KxleyStudios, 2023-06-07:
Hello! I wanted to talk to you about my **2ds XL,** I don't know how, but I always get **Arm11 (Core 0)** Errors
I don't know how, I was just wondering if you could help me. I have the latest version of Luma, but I may have updated luma after Nintendo sent out the update. I hope this is fixable, thank you!
[crash_dump_files.zip](https://github.com/LumaTeam/Luma3DS/files/11673245/crash_dump_files.zip)

jhjonesmo, 2023-07-01:
I can confirm the same issue. Crashing 2DS XL - Hyrule Edition 

[crash_dump_00000002.dmp.zip](https://github.com/LumaTeam/Luma3DS/files/11923926/crash_dump_00000002.dmp.zip)

PabloMK7, 2023-07-16:
The problem reported here seems to be related to corrupted storage, please follow the next steps:

If your game is a physical copy or a cartridge, take a look at [this post](https://github.com/skawo/GodMode9-with-Cartridge-Fixer). However, if your game is a digital copy, or if you have installed any updates or DLC to the game, your SD card may be corrupted. The following steps may help:

1) Remove the SD card from your 3DS and insert it into your computer. (Or any other device that is able to read SD cards.)

2) Copy all the contents from your SD card and paste them to a folder in your computer. This step is very important to prevent data loss.

3) Use your preferred formatting tool to format your SD card to FAT32 (even if it is already formatted to FAT32). If you need a guide read [this](https://3ds.hacks.guide/formatting-sd-(windows)).

4) Copy all the files you backed up in step 1 to your SD card again. This will rebuild the file system and remove any possible corruption.

5) Try to replicate the steps where SD corruption was causing issues. If you still are having issues, you will have to reinstall the application as the files were damaged (corruption won't damage them again). Make sure to backup the save file with any save manager before doing the reinstall!

6) If after doing the steps listed above you are still having issues, or corruption happens again in a short time, your SD card may be damaged or unreliable and has to be replaced. You can make sure by following [this guide](https://3ds.hacks.guide/h2testw-(windows)).

If the steps listed here don't help, please consider joining the [Nintendo Homebrew discord server](https://discord.gg/C29hYvh).

closed 2023-07-16 by PabloMK7