LumaTeam/Luma3DS #1905: [Feature Request] Induce Sleep Mode from Rosalina menu

spudpiggy, 2023-06-09:
My 3DS XL has a bad speaker ribbon, meaning that I have to keep the top screen at a certain angle or else it'll instantly turn off.
I really do hate this, I'll have to get it fixed at some point.
Anyway, one problem this causes is that I can't reliably close my console. I usually can, but I _need_ to save before just in case.
I know about magnets, but it's just kinda worrying seeing a magnet next to my 3DS, plus I don't want to have a magnet living in my other pocket.

I jsut wish there was a way, maybe either a Rosalina option or perhaps even a button combo, to enter and exit sleep mode whenever. This would make opening/closing my 3DS a bit less of a rush and there's probably a few other use cases.
Obviously it should be hidden for O2DS users due to the sleep switch existing.

Masamune3210, 2023-06-09:
Why would you get worried about magnets next to the 3ds? It's perfectly fine, especially considering the way the system knows to go into sleep mode is the speaker magnets in the top half being near the bottom half. That's why the magnet trick works in the first place

spudpiggy, 2023-06-10:
Magnets near electronics just looks concerning in general. Also, I shouldn't have to bring a magnet around with me to reliably induce sleep mode, i should just be able to do that through software in the first place.

Shadow322, 2023-06-10:
> My 3DS XL has a bad speaker ribbon, meaning that I have to keep the top screen at a certain angle or else it'll instantly turn off. I really do hate this, I'll have to get it fixed at some point. [...]

Replace the broken part not going to be a feature in Luma3DS